moggieuk/Happy-Hare #676: MMB-CANBUS-V2.0 with  EZ5160PRO  stel drivers

borislavstefanov, 2025-03-20:
Hello, I own the mmb-canbus-v2.0 board in question and 2 EZ5160PRO drivers arrived with it. I wonder if I need to configure anything in the drivers or the installation itself will do it for me, because I saw in the configuration files the settings are for 2209 drivers. Can anyone help with this problem? I've been trying to solve it for a long time.

moggieuk, 2025-03-24:
I believe the only thing that would be necessary is to change the tmc2209 references to tmc5160.  Happy Hare will look for this tmc type for capabilities like "touch" (stallguard) etc.

The only other thing to check if you are using stallguard are tmc specific settings like:
```
driver_SGTHRS: 75
```
They may not be the same for 5160

ningpj, 2025-03-24:
Depending on the mcu and 5160 sticks used, you may need to switch to spi and include software or hardware bus details in [tmc5160 stepper … ] blocks.

ningpj, 2025-03-26:
I dont have a btt mmu 1.x or 2.x so this is only a guess.  the tmc5160 ezdrivers are spi only from the spec's and appear to have a hardware spi bus (havent found refs to software spi).  as a minimum you will need to enable the spi_bus and set the sense_resistor to match tm5160 spec's. cs_pin, diag_pin from HH config.

```
[tmc5160 ez_driver1]
spi_bus: spi1
sense_resistor: 0.075
…

[tmc5160 ez_driver2]
spi_bus: spi1
sense_resistor: 0.075
…
```

cbknutson, 2025-04-05:
I just got my 5160 drivers working with my ERCF/Happy hare and BTT MMB (v1.1). Hope this helps anyone else searching Google.

# CBK - Attempt at 2209  steppers
[tmc5160 stepper_mmu_gear]
cs_pin: mmu:PA10
spi_speed: 4000000
spi_bus: spi1
#diag1_pin: PF0
interpolate: False
run_current: 0.6			# ERCF BOM NEMA14 motor
hold_current: 0.1
sense_resistor: 0.075
stealthchop_threshold: 0
#
# Uncomment two lines below if you have TMC and want the ability to use filament "touch" homing with gear stepper
#diag_pin: ^mmu:MMU_GEAR_DIAG		# Set to MCU pin connected to TMC DIAG pin for gear stepper
#driver_SGTHRS: 60			# 255 is most sensitive value, 0 is least sensitive

[stepper_mmu_gear]
step_pin: mmu:MMU_GEAR_STEP
dir_pin: !mmu:MMU_GEAR_DIR
enable_pin: !mmu:MMU_GEAR_ENABLE
rotation_distance: 22.7316868		# Bondtech 5mm Drive Gears. Overridden by 'mmu_gear_rotation_distance' in mmu_vars.cfg
gear_ratio: 80:20			        # E.g. ERCF 80:20, Tradrack 50:17
microsteps: 16 				# Recommend 16. Increase only if you "step compress" issues when syncing
full_steps_per_rotation: 200		# 200 for 1.8 degree, 400 for 0.9 degree

# CBK - Attempt at 2209  steppers
[tmc5160 stepper_mmu_gear]
cs_pin: mmu:PC7
spi_speed: 4000000
spi_bus: spi1
#diag1_pin: PF0
interpolate: False
run_current: 0.4
hold_current: 0.2
sense_resistor: 0.075
stealthchop_threshold: 100

[stepper_mmu_selector]
step_pin: mmu:MMU_SEL_STEP
dir_pin: !mmu:MMU_SEL_DIR
enable_pin: !mmu:MMU_SEL_ENABLE
rotation_distance: 40
microsteps: 16 				# Don't need high fidelity
full_steps_per_rotation: 200		# 200 for 1.8 degree, 400 for 0.9 degree
endstop_pin: ^mmu:MMU_SEL_ENDSTOP	# Selector microswitch
endstop_name: mmu_sel_home
# Uncomment this line only if default endstop above is using stallguard
#homing_retract_dist: 0

borislavstefanov, 2025-04-05:
for BT MMB CAN V2.0 will it be the same, can you also share the wiring method, thanks

borislavstefanov, 2025-04-05:
Ok, thank you  very much

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

In v4 you can pick TMC5160 / SPI as the driver type for each stepper in menuconfig, and it fills in the right sense resistor for you.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk