moggieuk/Happy-Hare #679: Gcode Preprocessing Failed

Gornerrrrr, 2025-04-02:
Hi! I am using HH+Boxturtle as a multi-color management module.
After configuring the **variable_automap_strategy: "closest_color"** property, I encountered an issue. Some files throw an error during printing: 
`Error during automapping When automapping with closest color, the tool material must be set. No gate has a filament matching the desired material (unknown). Available are : ['PLA', 'PLA', 'PLA', 'PLA'] Unable to find a suitable color for tool 3 (color: 000000)`

It seems this issue is caused by the **slicer_tool_map** not being assigned correctly. I then traced where it should be assigned, which is likely implemented through the **MMU_START_SETUP** calling **MMU_SLICER_TOOL_MAP**.
Therefore, I downloaded both the normal and error files for comparison:
```
**Correct**
;TYPE:Custom
MMU_START_SETUP INITIAL_TOOL=0 TOTAL_TOOLCHANGES=73 REFERENCED_TOOLS=0,1 TOOL_COLORS=FFFFFF,FF0000,000000,FFFFFF TOOL_TEMPS=220,220,220,220 TOOL_MATERIALS=PLA,PLA,PLA,PLA FILAMENT_NAMES="Generic PLA @System","Generic PLA @System","Generic PLA @System","Generic PLA @System" PURGE_VOLUMES=0.0,131.0,40.0,30.0,293.0,0.0,44.2,291.7,280.0,202.5,0.0,280.0,30.0,130.0,40.0,0.0
MMU_START_CHECK
PRINT_START EXTRUDER=220 BED=55
MMU_START_LOAD_INITIAL_TOOL
; Optionally add YOUR additional start logic (like purging) here to run just prior to start
SET_PRINT_STATS_INFO TOTAL_LAYER=80 ; For pause at layer functionality and better print stats
```
```
**Error**
;TYPE:Custom
MMU_START_SETUP INITIAL_TOOL=3 TOTAL_TOOLCHANGES=!total_toolchanges! REFERENCED_TOOLS=!referenced_tools! TOOL_COLORS=!colors! TOOL_TEMPS=!temperatures! TOOL_MATERIALS=!materials! FILAMENT_NAMES=!filament_names! PURGE_VOLUMES=!purge_volumes!
MMU_START_CHECK
PRINT_START EXTRUDER=230 BED=70
MMU_START_LOAD_INITIAL_TOOL
; Optionally add YOUR additional start logic (like purging) here to run just prior to start
SET_PRINT_STATS_INFO TOTAL_LAYER=437 ; For pause at layer functionality and better print stats
```
Upon comparison, I noticed that except for `INITIAL_TOOL`, none of the other values were replaced. 

What could be the reason for this, and how can it be fixed?

Gornerrrrr, 2025-04-03:
I used a consistent method for uploading files. After slicing with OrcaSlicer 2.3, I selected '**Upload and Print**,' and the entire process was operated within the same network segment.

ningpj, 2025-04-03:
You need to use Upload only to allow time for the moonraker preprocessor to parse and replace the HH tokens in the gcode file.  Depending on the size of the file and performance of your SD card it could take up to 30sec to rewrite the file.

Gornerrrrr, 2025-04-03:
> You need to use Upload only to allow time for the moonraker preprocessor to parse and replace the HH tokens in the gcode file. Depending on the size of the file and performance of your SD card it could take up to 30sec to rewrite the file.

Thanks I will try.

Gornerrrrr, 2025-04-07:
> You need to use Upload only to allow time for the moonraker preprocessor to parse and replace the HH tokens in the gcode file. Depending on the size of the file and performance of your SD card it could take up to 30sec to rewrite the file.

I have tried waiting for a while (10 minutes) after uploading before starting the print, but it didn’t work. The Gcode was not modified. Could you please tell me what other reasons might be causing this? I don’t think it’s related to the file size because I have other large print files that didn’t have any issues.

Gornerrrrr, 2025-04-11:
Found some failed information:
```
2025-04-10 08:36:03,276 [application.py:post()] - Processing Uploaded File: 3DBenchy_PLA_9h29m.gcode
2025-04-10 08:36:04,361 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Running MMU enhanced version of metadata

2025-04-10 08:36:04,373 [shell_command.py:pipe_data_received()] - INFO:metadata:Object Processing is disabled

2025-04-10 08:36:04,373 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Pre-processing file: /home/orangepi/printer_data/gcodes/3DBenchy_PLA_9h29m.gcode

2025-04-10 08:36:18,254 [shell_command.py:pipe_data_received()] - INFO:metadata:Reading placeholders took 13.88s. Detected gcode by slicer: OrcaSlicer

2025-04-10 08:36:23,784 [shell_command.py:_check_proc_success()] - Command (/home/orangepi/moonraker-env/bin/python /home/orangepi/moonraker/moonraker/components/mmu_server.py -m -n -c /tmp/metacfg-6_b26j54.json) cancelled
2025-04-10 08:36:23,786 [file_manager.py:_process_metadata_update()] - Error running extract_metadata.py
Traceback (most recent call last):
  File "/home/orangepi/moonraker/moonraker/components/file_manager/file_manager.py", line 2566, in _process_metadata_update
    await self._run_extract_metadata(fname, ufp_path)
  File "/home/orangepi/moonraker/moonraker/components/file_manager/file_manager.py", line 2621, in _run_extract_metadata
    raise self.server.error("Extract Metadata returned with error")
moonraker.utils.exceptions.ServerError: Extract Metadata returned with error
2025-04-10 08:36:24,382 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Running MMU enhanced version of metadata

2025-04-10 08:36:24,394 [shell_command.py:pipe_data_received()] - INFO:metadata:Object Processing is disabled

2025-04-10 08:36:24,395 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Pre-processing file: /home/orangepi/printer_data/gcodes/3DBenchy_PLA_9h29m.gcode

2025-04-10 08:36:38,257 [shell_command.py:pipe_data_received()] - INFO:metadata:Reading placeholders took 13.86s. Detected gcode by slicer: OrcaSlicer

2025-04-10 08:36:43,819 [shell_command.py:_check_proc_success()] - Command (/home/orangepi/moonraker-env/bin/python /home/orangepi/moonraker/moonraker/components/mmu_server.py -m -n -c /tmp/metacfg-g422stot.json) cancelled
2025-04-10 08:36:43,822 [file_manager.py:_process_metadata_update()] - Error running extract_metadata.py
Traceback (most recent call last):
  File "/home/orangepi/moonraker/moonraker/components/file_manager/file_manager.py", line 2566, in _process_metadata_update
    await self._run_extract_metadata(fname, ufp_path)
  File "/home/orangepi/moonraker/moonraker/components/file_manager/file_manager.py", line 2621, in _run_extract_metadata
    raise self.server.error("Extract Metadata returned with error")
moonraker.utils.exceptions.ServerError: Extract Metadata returned with error
2025-04-10 08:36:44,394 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Running MMU enhanced version of metadata

2025-04-10 08:36:44,410 [shell_command.py:pipe_data_received()] - INFO:metadata:Object Processing is disabled

2025-04-10 08:36:44,411 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Pre-processing file: /home/orangepi/printer_data/gcodes/3DBenchy_PLA_9h29m.gcode

2025-04-10 08:36:58,125 [shell_command.py:pipe_data_received()] - INFO:metadata:Reading placeholders took 13.71s. Detected gcode by slicer: OrcaSlicer

2025-04-10 08:37:03,855 [shell_command.py:_check_proc_success()] - Command (/home/orangepi/moonraker-env/bin/python /home/orangepi/moonraker/moonraker/components/mmu_server.py -m -n -c /tmp/metacfg-y2aqst60.json) cancelled
2025-04-10 08:37:03,858 [file_manager.py:_process_metadata_update()] - Error running extract_metadata.py
Traceback (most recent call last):
  File "/home/orangepi/moonraker/moonraker/components/file_manager/file_manager.py", line 2566, in _process_metadata_update
    await self._run_extract_metadata(fname, ufp_path)
  File "/home/orangepi/moonraker/moonraker/components/file_manager/file_manager.py", line 2621, in _run_extract_metadata
    raise self.server.error("Extract Metadata returned with error")
moonraker.utils.exceptions.ServerError: Extract Metadata returned with error
2025-04-10 08:37:03,858 [file_manager.py:_process_metadata_update()] - Unable to extract medatadata from file: 3DBenchy_PLA_9h29m.gcode
2025-04-10 08:37:03,859 [klippy_apis.py:start_print()] - Requesting Job Start, filename = 3DBenchy_PLA_9h29m.gcode
2025-04-10 08:37:03,889 [application.py:log_request()] - 201 POST /api/files/local (192.168.3.159) [_TRUSTED_USER_] 119187.17ms
2025-04-10 08:37:03,958 [job_state.py:_status_update()] - Job Started: 3DBenchy_PLA_9h29m.gcode
2025-04-10 08:37:59,177 [websockets.py:on_close()] - Websocket Closed: ID: 547817711952 Close Code: 1001, Close Reason: None, Pong Time Elapsed: 0.58
2025-04-10 08:37:59,333 [application.py:log_request()] - 101 GET /websocket (192.168.3.159) [_TRUSTED_USER_] 8.04ms
2025-04-10 08:37:59,335 [websockets.py:open()] - Websocket Opened: ID: 547817706768, Proxied: True, User Agent: BBL-Slicer/v01.10.01.50 (light) Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/107.0.0.0 Safari/537.36 Edg/107.0.1418.52, Host Name: 192.168.3.71
2025-04-10 08:37:59,351 [websockets.py:_handle_identify()] - Websocket 547817706768 Client Identified - Name: mainsail, Version: 2.13.2, Type: web
[... cut ...]
2025-04-10 08:50:08,480 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Writing MMU placeholders,Inserting next position to tool changes took 1.08s

2025-04-10 08:50:08,602 [shell_command.py:pipe_data_received()] - INFO:metadata:Processor preprocess_cancellation is disabled

2025-04-10 08:50:08,923 [shell_command.py:_check_proc_success()] - Command (/home/orangepi/moonraker-env/bin/python /home/orangepi/moonraker/moonraker/components/mmu_server.py -m -n -c /tmp/metacfg-ynn44obw.json) successfully finished
2025-04-10 08:50:08,933 [application.py:log_request()] - 201 POST /api/files/local (192.168.3.159) [_TRUSTED_USER_] 13521.86ms
```

closed 2026-10-21 by moggieuk