moggieuk/Happy-Hare #680: CALIBRATE_BOWDEN(Auto) Does not work. Same as Issue#441

AlexTOOT, 2025-04-02:
Running v3.1.0-162-g3d367ce, I have got the exact same issue with this one: https://github.com/moggieuk/Happy-Hare/pull/441#issue-2546820470.

In bowden calibration, loading process is fine, but in backing up to gate, it unloads only same amount with "gate_homing_max". Don't know why that happens.

Here is what shows on terminal.
$ MMU_CALIBRATE_BOWDEN BOWDEN_LENGTH=1000
22:54:44
// Calibrating bowden length for gate 0 (automatic method) using ENCODER as gate reference point
22:55:06
// Filament homed to extruder after 941.3mm movement
// (encoder measured 930.0mm)
22:55:11
!! MMU issue: Calibration of bowden length on gate 0 failed. Aborting because:
Failed to unload gate because did not clear the encoder after moving 100.0mm(as my gate_homing_max is 100)

But if I set gate_homing_max to something big, like 1000, another error pops up
after loading and before unloading, it says:
!! MMU issue: Calibration of bowden length on gate 0 failed. Aborting because:
The extruder sensor triggered before homing. Check filament and sensor operation

Any ideas? or is log needed?

AlexTOOT, 2025-04-11:
Downgraded to v2.7.3，problem fixed. I don't know why.

moggieuk, 2025-04-29:
I believe this is/was happening because the pre-gate sensor for the selected gate did not detect filament and this caused the bowden unload move to be skipped.   I pushed an update a couple of days ago to simply warning of this failure but to still process with the bowden load.

AlexTOOT, 2025-05-02:
> I really need a log to be able to help, ideally with TRACE level turned on:
Sorry, no can do, I already downgraded to V2.7, which works fine.

> I believe this is/was happening because the pre-gate sensor ...
Yes, I think that would be the problem, I unplugged the pre-gate sensors a long time ago and didn't comment them out in config. 
And I've updated back to 3.1 a few days ago, the problem is gone. So I think the patch works.

Thanks for your time and awesome work.

closed 2025-05-02 by AlexTOOT