moggieuk/Happy-Hare #682: How to Eject unused Gates material in Type B MMU?

Gornerrrrr, 2025-04-08:
I’m currently using HappyHare+Boxtrutle. When T1 with Gate1 for printing, but I’d like to change the material on Gate3 without interrupting the printing process. Currently, the system requires me to Pause Print + Unload T1 + Select Gate3 just to change the material on Gate3. 

I suspect that in the HappyHare structure, Unload material and switching Gates is necessary, but in Boxturtle, where each channel is independently controlled, this doesn’t seem necessary. Is it possible to implement such a quick operation? (Like Pause + Eject Gate3)

If supporting such an operation is possible, it would make the process much more convenient and efficient. 
Thank you for your response!

igiannakas, 2025-04-09:
Right now I don’t think you can do this. It’s a great idea and much needed for type b MMus as the gates can eject, no problem . 

However you can’t load new filament to the empty gate as it needs to home to the gate sensor and that won’t be possible while the gate sensor is occupied with filament from another gate.

Gornerrrrr, 2025-04-10:
> Right now I don’t think you can do this. It’s a great idea and much needed for type b MMus as the gates can eject, no problem .
>
> However you can’t load new filament to the empty gate as it needs to [...]

Thanks for your reply!
When I insert new filament, the MMU loads the filament by detecting the PRE_GATE_SENSOR until the POST_GEAR is triggered. The occupied GATE_SENSOR doesn’t seem to be necessary during the preloading process.

Gornerrrrr, 2025-04-10:
I tried unloading Gate1 while loading Gate3. `MMU_EJECT GATE=1 FORCE=1`

It‘s can be work:
1. Toolhead to Forming tip...
2. Gate1 was successfully ejected
3. MMU issue: 
```
Filament eject for gate 1 failed: Unload sequence failed because:
Failed to reach extruder entry sensor after moving 81.0mm
```

It might just require some minor adjustments to achieve this quick unloading and material change.

Gornerrrrr, 2025-05-06:
> Hi [@Gornerrrrr](https://github.com/Gornerrrrr) you should be able to do what you want when filament is loaded but not when printing (technically possible but very easy to mess up the print in progress so currently disabled).
> 
> Was/is the limitation with the Mainsail/Fluidd UI? (if so then this has been fixed) or did it occur with the command line:
> 
> ```
> MMU_EJECT GATE=x
> 
> MMU_PRELOAD GATE=x
> ```
> 
> The preloading case is possible if you are gate homing to the "mmu_gear" sensor or have a "mmu_gear" sensor configured (the default).

Thanks reply @moggieuk. I'm use Mainsail and command line.
In Pausing, Im try **Eject Gate[3]** when Gate[0] in main channel:
```
10:53:42 MMU_EJECT GATE=3
10:53:42 Parking toolhead at (x:n/a, y:n/a, z:5.2) for unload operation
10:53:42 Parking toolhead at (x:150.0, y:10.0, z:5.2) for force_park operation
10:53:43 Unloading filament...
10:53:43 [T0] < (*) <<<<<<<<<< (*) << [Ex << (*) << Nz] LOADED 0.0mm
10:53:43 Forming tip...
10:53:46 [T0] < (*) <<<<<<<<<< (*) << [Ex << (*) <. Nz] -70.0mm
10:53:46 Parking toolhead at (x:30.0, y:10.0, z:5.2) for force_park operation
10:53:54 MMU issue detected whilst printer is paused
Reason: Filament eject for gate 3 failed: Unload sequence failed because:
Failed to reach extruder entry sensor after moving 78.0mm
10:53:54 Attempting to recover filament position...
10:53:54 [T0] < (*) <<<<<<<<<< (*) << [Ex << (*) << Nz] LOADED -148.0mm
```
It can be noticed that there are unnecessary actions of **Unload Tool[0]** from the main channel, and failure alarms are detected.
```
11:02:37 MMU_EJECT GATE=0
11:02:39 Parking toolhead at (x:150.0, y:10.0, z:5.2) for force_park operation
11:02:40 Unloading filament...
11:02:40 [T0] < (*) <<<<<<<<<< (*) << [Ex << (*) << Nz] LOADED 0.0mm
11:02:40 Forming tip...
11:02:43 [T0] < (*) <<<<<<<<<< (*) << [Ex << (*) <. Nz] -70.0mm
11:02:43 Parking toolhead at (x:30.0, y:10.0, z:5.2) for force_park operation
11:02:48 [T0] < (*) <<<<<<<<<<|(e) .. [Ex .. (t) .. Nz] -123.1mm
11:03:00 Disabled extruder heater
11:03:01 [T0] < (*) <<........ (e) .. [Ex .. (t) .. Nz] -1460.6mm
11:03:02 [T0] <|(g) .......... (e) .. [Ex .. (t) .. Nz] -1473.1mm
11:03:02 [T0] < (g) .......... (e) .. [Ex .. (t) .. Nz] UNLOADED -1533.1mm
11:03:03 Unload of -1533.1mm filament successful
11:03:04 Ejecting filament out of current gate
11:03:04 Ejecting...
11:03:07 The filament in gate 0 can be removed
11:05:48 MMU_EJECT GATE=3
11:05:48 Ejecting filament out of current gate
11:05:48 Ejecting...
11:05:48 The filament in gate 3 can be removed
```
The unnecessary **Eject Gate [0]** must be executed first, and then the **Eject Gate [3]** can be executed normally.

igiannakas, 2025-06-05:
Bumping to keep open. That’s a nice QOL thing for type B MMus

Gornerrrrr, 2025-07-10:
I attempted to modify the code myself and it is currently running smoothly.
Just need to modify the def cmd_MMU_EJECT(self, gcmd) function in ./extras/mmu/mmu.py.
```
    def cmd_MMU_EJECT(self, gcmd):
        self.log_to_file(gcmd.get_commandline())
        if self.check_if_disabled(): return
        gate = gcmd.get_int('GATE', self.gate_selected, minval=0, maxval=self.num_gates - 1)
        force = bool(gcmd.get_int('FORCE', 0, minval=0, maxval=1))
        if self.check_if_not_calibrated(self.CALIBRATED_ESSENTIAL, check_gates=[gate]): return
        self._fix_started_state()

        # Type B: Skip the "print head removal" detection
        skip_unload = False
        can_crossload = self.mmu_machine.multigear and self.sensor_manager.has_gate_sensor(self.SENSOR_GEAR_PREFIX, gate)
        if gate != self.gate_selected:
            if can_crossload:
                self.log_always("[EJECT_TEST]: cross_gate_eject detected → skip_unload=True (gate=%d, current_gate=%d)" %
                (gate, self.gate_selected))
                skip_unload = True
            elif self.check_if_loaded():
                return

        # Determine if eject_from_gate is necessary
        in_bypass = self.gate_selected == self.TOOL_GATE_BYPASS
        extruder_only = bool(gcmd.get_int('EXTRUDER_ONLY', 0, minval=0, maxval=1)) or in_bypass
        can_eject_from_gate = (
            not extruder_only
            and (
                self.mmu_machine.multigear and gate != self.gate_selected
                or self.filament_pos == self.FILAMENT_POS_UNLOADED
                or force
            )
        )
        self.log_always("[EJECT_TEST]: can_eject_from_gate= !extruder_only(%s) or force(%s) -> %s" %
                (extruder_only, force, can_eject_from_gate))
        if not can_eject_from_gate and self.filament_pos == self.FILAMENT_POS_UNLOADED:
            self.log_always("Filament not loaded")
            return

        try:
            with self.wrap_sync_gear_to_extruder():
                with self._wrap_suspend_runout(): # Don't want runout accidently triggering during filament eject
                    with self.wrap_action(self.ACTION_EJECTING, gate):
                        current_gate = self.gate_selected
                        self.select_gate(gate)
                        if not skip_unload:
                          self._mmu_unload_eject(gcmd)
                        if can_eject_from_gate:
                            self.log_always("Ejecting filament out of %s" % ("current gate" if gate == self.gate_selected else "gate %d" % gate))
                            self._eject_from_gate(gate, current_gate)
                        # If necessary or easy restore previous gate
                        if self.is_in_print() or self.mmu_machine.multigear:
                            self.select_gate(current_gate)
                        else:
                            self._initialize_encoder() # Encoder 0000
                            self._auto_filament_grip()

                        self._persist_swap_statistics()

        except MmuError as ee:
            self.handle_mmu_error("Filament eject for gate %d failed: %s" % (gate, str(ee)))
```