moggieuk/Happy-Hare #683: Motor sync is not enabled by default

fineemb, 2025-04-09:
During the first print after powering on, if the current channel has filaments loaded, synchronization will not start. Synchronization will only start correctly after the channel has been changed at least once.

In addition, the parameter` sync_to_extruder: 1` seems to be invalid. When it is turned on, an error message will be displayed: Operation not possible. `MMU design requires continuous gear/extruder syncing`

igiannakas, 2025-04-09:
Are you using an ERCF or a Type B mmu (ie with a stepper on each gate?)

fineemb, 2025-04-09:
Type B mmu

[image]

moggieuk, 2025-04-29:
Sorry for the delay in getting to this...
The 'sync_to_extruder' is redundant yes (because of shared use of config file) because it will always be forced true, but it shouldn't issue any error if it is set to 0.   If you run the `MMU_SYNC_GEAR_MOTOR` it will report that error.

On the startup case, that could be a bug. I'll take a look...  Does it still occur in HH v3.2 ?

bickus, 2025-07-07:
Here's the fix I've done for myself. Please ignore the incorrect logging, Python is not my favourite :)
[](https://github.com/bickus/Happy-Hare/commit/d29aa77c21feb6c6323cd3a8435f2205efbbae6c)

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

In v4 an MMU with a gear per gate is always synced whenever filament is in the extruder, including at print start with a gate already loaded.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk