moggieuk/Happy-Hare #684: Spool_ID is not retrieved, receive error message after switching on

MDegelmann, 2025-04-09:
When I start the printer I get this error message.

[image]

No settings are retrieved either.

[image]
[image]

Where is my mistake :(

igiannakas, 2025-04-09:
Post your moonraker config file contents please. Spoolman needs to be set up there too. For example like so:

```
[spoolman]
server: http://127.0.0.1:7912
sync_rate: 5
```

MDegelmann, 2025-04-09:
I have set up Spoolman itself but it doesn't work properly

`[server]
host: 0.0.0.0
port: 7125
klippy_uds_address: /home/biqu/printer_data/comms/klippy.sock

[authorization]
trusted_clients:
    192.168.0.0/16
    10.0.0.0/8
    127.0.0.0/8
    169.254.0.0/16
    172.16.0.0/12
    192.168.0.0/16
    FE80::/10
    ::1/128
cors_domains:
    *.lan
    *.local
    *://localhost
    *://localhost:*
    *://my.mainsail.xyz
    *://app.fluidd.xyz

[octoprint_compat]

[history]

[spoolman]
server: http://192.168.0.219:7912
#   URL to the Spoolman instance. This parameter must be provided.
sync_rate: 5
#   The interval, in seconds, between sync requests with the
#   Spoolman server.  The default is 5

[update_manager]
channel: dev
refresh_interval: 168

[update_manager mainsail]
type: web
channel: stable
repo: mainsail-crew/mainsail
path: ~/mainsail

[update_manager mainsail-config]
managed_services: klipper
origin: https://github.com/mainsail-crew/mainsail-config.git
path: /home/biqu/mainsail-config
primary_branch: master
type: git_repo

[update_manager KlipperScreen]
type: git_repo
path: ~/KlipperScreen
origin: https://github.com/moggieuk/KlipperScreen-Happy-Hare-Edition.git
env: ~/.KlipperScreen-env/bin/python
requirements: scripts/KlipperScreen-requirements.txt
install_script: scripts/KlipperScreen-install.sh
managed_services: KlipperScreen

[update_manager BDsensor]
type: git_repo
primary_branch: new
channel: dev
path: ~/Bed_Distance_sensor
origin: https://github.com/markniu/Bed_Distance_sensor.git
install_script: ./klipper/install_BDsensor.sh
is_system_service: False
managed_services: klipper
info_tags:
  desc=Bed Distance Sensor

[update_manager Klippain-ShakeTune]
type: git_repo
origin: https://github.com/Frix-x/klippain-shaketune.git
path: ~/klippain_shaketune
virtualenv: /home/biqu/klippy-env
requirements: requirements.txt
system_dependencies: system-dependencies.json
primary_branch: main
managed_services: klipper

[update_manager happy-hare]
type: git_repo
path: ~/Happy-Hare
origin: https://github.com/moggieuk/Happy-Hare.git
primary_branch: main
managed_services: klipper

[mmu_server]
enable_file_preprocessor: True
enable_toolchange_next_pos: True
update_spoolman_location: True

`

igiannakas, 2025-04-09:
When you go to this IP do you see the spoolman web interface? http://192.168.0.219

MDegelmann, 2025-04-09:
No the mainsail

MDegelmann, 2025-04-09:
No problem
Yes, that's the address of the printer on which Spoolman is installed.

igiannakas, 2025-04-09:
Can you try this IP in your browser? http://192.168.0.219:7912/

Does it open the spoolman interface?

Also may be worth changing it to server: http://127.0.0.1:7912

Which is the local host address. Trying to rule out networking issues first.

Also I presume you’ve rebooted the printer, right?

MDegelmann, 2025-04-09:
Both 192.168.0.219:7912 opens Spoolman.
Mainsail connects at 192.168.0.219:7912 and localhost:7912. It connects, but I get these errors.

[image]

igiannakas, 2025-04-09:
Is your moon raker instance up to date? That’s about the last thing I can think off.

Maybe also post some logs - moonraker log should help

MDegelmann, 2025-04-09:
Is everything updated

-------------------- Log Start | Wed Apr  9 20:15:19 2025 --------------------
platform: Linux-6.6.66-vendor-sunxi64-aarch64-with-glibc2.36
data_path: /home/biqu/printer_data
is_default_data_path: False
config_file: /home/biqu/printer_data/config/moonraker.conf
backup_config: /home/biqu/printer_data/config/.moonraker.conf.bkp
startup_warnings: []
verbose: False
debug: False
asyncio_debug: False
is_backup_config: False
is_python_package: False
instance_uuid: be06a45c2bb944fc8fe0683231abeb01
unix_socket_path: /home/biqu/printer_data/comms/moonraker.sock
structured_logging: False
software_version: v0.9.3-72-g7cdcca3
git_branch: master
git_remote: origin
git_repo_url: https://github.com/Arksine/moonraker
modified_files: []
unofficial_components: []
log_file: /home/biqu/printer_data/logs/moonraker.log
python_version: 3.11.2 (main, Nov 30 2024, 21:22:50) [GCC 12.2.0]
launch_args: /home/biqu/moonraker-env/bin/python /home/biqu/moonraker/moonraker/moonraker.py -d /home/biqu/printer_data
msgspec_enabled: True
uvloop_enabled: True

#################### Moonraker Configuration ####################

[server]
host = 0.0.0.0
port = 7125
klippy_uds_address = /home/biqu/printer_data/comms/klippy.sock

[authorization]
trusted_clients = 
	192.168.0.0/16
	10.0.0.0/8
	127.0.0.0/8
	169.254.0.0/16
	172.16.0.0/12
	192.168.0.0/16
	FE80::/10
	::1/128
cors_domains = 
	*.lan
	*.local
	*://localhost
	*://localhost:*
	*://my.mainsail.xyz
	*://app.fluidd.xyz

[octoprint_compat]

[history]

[spoolman]
server = http://192.168.0.219:7912
sync_rate = 5

[update_manager]
channel = dev
refresh_interval = 168

[update_manager mainsail]
type = web
channel = stable
repo = mainsail-crew/mainsail
[... 680 lines cut ...]
2025-04-09 22:17:49,004 [server.py:add_log_rollover_item()] - Connected to Spoolman Spool Manager
2025-04-09 22:17:49,143 [websockets.py:_handle_identify()] - Websocket 281473507854160 Client Identified - Name: KlipperScreen, Version: v3.0.0-22-g5561b48f, Type: display
2025-04-09 22:17:49,212 [application.py:log_request()] - 404 GET /printer/info (127.0.0.1) [_TRUSTED_USER_] 24.60ms
2025-04-09 22:17:49,294 [mmu_server.py:_get_extra_fields()] - Extra fields for spool found
2025-04-09 22:17:49,294 [mmu_server.py:_build_spool_location_cache()] - Building spool location cache from Spoolman db
2025-04-09 22:17:49,321 [spoolman.py:_check_spool_deleted()] - Spool ID 0 not found, setting to None
2025-04-09 22:17:49,322 [spoolman.py:set_active_spool()] - Setting active spool to: None
2025-04-09 22:17:53,912 [application.py:log_request()] - 404 GET /printer/info (127.0.0.1) [_TRUSTED_USER_] 4.62ms
2025-04-09 22:17:56,487 [server.py:add_log_rollover_item()] - Klipper Version: v0.12.0-490-g5493bdfb4-dirty
2025-04-09 22:18:00,126 [file_manager.py:get_file_list()] - Updating File List <gcodes>...
2025-04-09 22:18:00,353 [klippy_connection.py:_request_initial_subscriptions()] - Webhooks Subscribed
2025-04-09 22:18:00,356 [klippy_connection.py:_request_initial_subscriptions()] - GCode Output Subscribed
2025-04-09 22:18:00,605 [job_state.py:_handle_started()] - Job state initialized: standby
2025-04-09 22:18:00,889 [klippy_connection.py:_check_ready()] - Klippy ready
2025-04-09 22:18:01,352 [data_store.py:_init_sensors()] - Configuring available sensors: ['temperature_sensor CB1', 'temperature_sensor Manta', 'heater_bed', 'extruder']
2025-04-09 22:18:03,349 [mmu_server.py:_init_mmu_backend()] - MMU backend present: True
2025-04-09 22:18:03,349 [mmu_server.py:_init_mmu_backend()] - MMU backend enabled: True
2025-04-09 22:18:03,350 [mmu_server.py:_initialize_mmu()] - MMU num_gates: 8
2025-04-09 22:18:41,602 [authorization.py:_check_trusted_connection()] - Trusted Connection Detected, IP: 192.168.0.174
2025-04-09 22:18:41,604 [application.py:log_request()] - 101 GET /websocket (192.168.0.174) [_TRUSTED_USER_] 4.22ms
2025-04-09 22:18:41,605 [websockets.py:open()] - Websocket Opened: ID: 281473507738832, Proxied: True, User Agent: BBL-Slicer/v01.10.01.50 (dark) Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/107.0.0.0 Safari/537.36 Edg/107.0.1418.52, Host Name: 192.168.0.219
2025-04-09 22:18:41,609 [websockets.py:_handle_identify()] - Websocket 281473507738832 Client Identified - Name: mainsail, Version: 2.13.2, Type: web
2025-04-09 22:18:47,061 [application.py:log_request()] - 101 GET /websocket (192.168.0.174) [_TRUSTED_USER_] 3.29ms
2025-04-09 22:18:47,062 [websockets.py:open()] - Websocket Opened: ID: 281473505139600, Proxied: True, User Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/134.0.0.0 Safari/537.36, Host Name: 192.168.0.219
2025-04-09 22:18:47,081 [websockets.py:_handle_identify()] - Websocket 281473505139600 Client Identified - Name: mainsail, Version: 2.13.2, Type: web

igiannakas, 2025-04-09:
The log points to the same issue as this. https://github.com/Donkie/Spoolman/issues/504

Suggest you connect there and report your issue too

MDegelmann, 2025-04-13:
Spoolman runs without MMU without any problems. Push mode also works without any problems, but pull mode has problems. That's why I see the error Happy

MDegelmann, 2025-04-13:
> When I start the printer I get this error message.
> 
> [image]
> 
> No settings are retrieved either.
> 
> [image] [image]
> 
> Where is my mistake :(

moggieuk, 2025-04-29:
@MDegelmann sorry for the delay in looking at this.  So you are saying that "push" mode works fine, but "pull" mode has the error?  That would be very strange because the "remote methods" desribed in the error message are all registered together.

I thought you problem was because the Happy Hare modified Moonraker was not running. That can be fixed by simply running this again.
```
cd ~/Happy-Hare
./install.sh
```

BTW It is highly likely that you want "push" mode .   The "pull" mode is really designed for print farms and means that you HAVE to configure what filament are on what printers/mmu in spoolman.  Typically is much easier to do so on the printer the MMU is attached to and have it update spoolman (for visibility)

closed 2026-10-21 by moggieuk