moggieuk/Happy-Hare #694: Gatemap temperature should be temperature of 1st filament

alidaf, 2025-04-22:
I'm confused by the gatemap temperature defaulting to 200°C when HH already has the material information from the slicer. Is there a way to change this behaviour?

moggieuk, 2025-04-22:
Can you tell me what version of HH you are running and also attach a `mmu.log`..

alidaf, 2025-04-22:
I've just upgraded to 3.2 after the Voron has been out of use for a few months. I'm still catching up after an abortive attempt with ERCF V1, and now trying ERCF V2.

[mmu.log](https://github.com/user-attachments/files/19848581/mmu.log)

moggieuk, 2025-04-28:
Ok. I assume you are referring to the logic that results in this part of the log:
```yml
11:36:56 > MMU_CHANGE_TOOL STANDALONE=1 RESTORE=0 TOOL=0
11:36:56   DEBUG: Tool change initiated with standalone MMU tip forming and standalone MMU purging
11:36:56 Tool change requested: T0
11:36:56   DEBUG: Saving toolhead gcode state and position (X:175.0 Y:175.0 Z:10.0 E:0.0) for toolchange
11:36:56 Saving toolhead position (x:175.0, y:175.0, z:10.0)
11:36:56   DEBUG: Extruder is not hot enough to retract
11:36:56 Parking toolhead at (x:50.0, y:50.0, z:10.4) for toolchange operation
11:36:56   DEBUG: Loading tool T0...
11:36:58 Loading filament...
11:36:58 [T0] > En ......... (e) .. [Ex .. (t) .. Nz] UNLOADED 0.0mm (e:0.0mm)
11:37:00 > MMU_STATS COUNTER=servo_down INCR=1
11:37:01 [T0] > En >........ (e) .. [Ex .. (t) .. Nz] 70.0mm (e:56.8mm)
11:37:01   DEBUG: Loading bowden tube
11:37:09 [T0] > En >>>>>>>>> (e) .. [Ex .. (t) .. Nz] 763.2mm (e:777.6mm)
11:37:09   DEBUG: Loading filament into extruder
11:37:09 Warning: Automatically heating extruder to gatemap temp (200.0°C)
11:37:09 Waiting for extruder to reach target (gatemap) temperature: 200.0°C
```
If so, you can see that to 200 degrees is being used from the "gatemap" (which defaults to the HH min temp unless overriden for that particular gate/filament).  The reason is that the initial tool load is outside of the slicer control - normally in a print the slicer will set temperature and so HH does nothing.   I don't really understand why slicers don't include the initial tool change (Txxx) command including setting extruder temp...

Anyway, I get you point that since, in your case, the temp has been parsed from the slicer produced gcode it would make more sense to use that rather than the standalone "gatemap" temp.   HH doesn't currently do this but you can workaround by setting the correct temp in the gate map with, eg.
```yml
MMU_GATE_MAP GATE=0 TEMP=230
```

I'll add to the to-do list to add the functionality you expect because it does make more sense to use slicer info if that is available.

moggieuk, 2026-10-25:
Closing because I believe this has been resolved.  Also the upcoming v4.2.0 release has much more sophisticated temperature setting based including understand the material type left in the nozzle from previous toolchange / print.

closed 2026-10-25 by moggieuk