moggieuk/Happy-Hare #696: Purging on print start and unwanted un-retracting

bakfiets2, 2025-04-23:
After the last update. Every print start. Starts with a purge of 29.4mm.  Also on every toolchange. The printer Un-retracts 2mm of filament. This results in a messy wipetower. Both things did not happen on 3.1. 

In the start Gcode i have a own purge line configured.

Wich setting became active after the update?

Matze9876, 2025-04-23:
The same for me.
since the update happy hare has an inexplicable purge of 107.5mm

[image]

ningpj, 2025-04-23:
there is a new ``_mmu_purge`` macro which likely defaulted to on in 3.2 e.g ``purge_macro: _MMU_PURGE``
you cant disable it by blanking ``purge_macro:`` in mmu_parameters.cfg or replace it with your own purge macro.

SwiftingSpeed, 2025-04-24:
It seems like the default was ment to be blank in the initial commit for the new purge logic: https://github.com/moggieuk/Happy-Hare/commit/43fbb6860b62008624c5bb0eb26d0feb850cd53f

However "_MMU_PURGE" looks like it slipped in during a e-spooler change: https://github.com/moggieuk/Happy-Hare/commit/aabb8b2db375c6aa5158c5576f1f313817f385bc

bakfiets2, 2025-04-24:
Even after blanking out the macro. The Un-Retract keeps happening.

bakfiets2, 2025-05-08:
Still no fix?. Happy hare becomes un-usable with also the tip forming servo problems. Wipe tower is a big mess.

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

In v4 the purge method is a menuconfig choice (pick 'None / slicer only' to stop Happy Hare purging), and the 2mm un-retract is the last number in the toolchange park setting, which you can set to 0.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk