moggieuk/Happy-Hare #697: Tip forming servo

bakfiets2, 2025-04-24:
During tip forming in the extruder. I don't use the sync gear motor function. But the servo stays down during tip forming. Gearmotor does not run. So filament is grinding in the MMU

moggieuk, 2025-04-27:
Oh, this must be a new regression in v3.2.  Can you supply your `mmu.log` - I'm traveling so can't test on my printer but can probably see the omission from the log...

bakfiets2, 2025-04-27:
Yes. Version is 3.2. This started after the update. I wil do a test print to generate a mmu log file.

MMU_SYNC_GEAR_MOTOR SYNC=1 is not in my start gcode. It is disabled in the config.

bakfiets2, 2025-04-29:
Here is the log file.

Also new is on unloading fail with about 10mm of unload, it gives a error not enough unloaded. Still swaps to another tool. Crashing the selector in the still loaded filament. 

[mmu(3).log](https://github.com/user-attachments/files/19965660/mmu.3.log)

bakfiets2, 2025-05-04:
What is the status? MMU is not usable at this moment because of the servo