moggieuk/Happy-Hare #698: 3.2 mmu_macro_vars update issue

Sparkss, 2025-04-26:
The update automation appears to have had an error converting the variable_park_toolchange parameter from 3.1 to 3.2. It caused a move out of range error during a tool change. I can provide the full mmu.log but the cause of the MOOR event is obvious and was resolved when I manually updated the entry. Here are the pertinent error entries from the log.

10:27:00 Tool change requested: T0
10:27:00 Error running _MMU_PARK: Move out of range: 149.432 138.884 260.050 [-2.000]
10:27:01 Load sequence failed because:
         Error running _MMU_PRE_LOAD: Move out of range: 150.000 128.900 260.050 [-2.000].
         Occured when changing tool: > T0. Retrying...

The original line
variable_park_toolchange        : -1, -1, -1, 10, 2	; x,y,z-hop,z_hop_ramp,retract for "toolchange" operations (toolchange,load,unload)

The converted line
variable_park_toolchange        : -999, -1, -999, 10, 2	; x,y,z-hop,z_hop_ramp,retract for "toolchange" operations (toolchange,load,unload)

The 'fixed' line
variable_park_toolchange        : -999, -999, 1, 10, 2	; x,y,z-hop,z_hop_ramp,retract for "toolchange" operations (toolchange,load,unload)

## I realized that I had the z-hop set to -1 in my original cfg when it should have been 0 to disable it. After thinking about it I bumped it to 1. This is unrelated to the variable update logic

moggieuk, 2025-04-29:
Thank you

```
The original line
variable_park_toolchange : -1, -1, -1, 10, 2 ; x,y,z-hop,z_hop_ramp,retract for "toolchange" operations (toolchange,load,unload)

The converted line
variable_park_toolchange : -999, -1, -999, 10, 2 ; x,y,z-hop,z_hop_ramp,retract for "toolchange" operations (toolchange,load,unload)
```

If this is what you observed then it is definitely a bug.  It should have converted to:
```
variable_park_toolchange : -999, -999, -1, 10, 2 
```

I will check on this...

moggieuk, 2025-04-29:
Fix has been pushed. Thank you!