moggieuk/Happy-Hare #699: Blobifier edge case move out of range error

Sparkss, 2025-04-26:
I kept running into a MOOR (Move Out Of Range) error when purging. After putting in tons of debugging I traced it to this part of the blobifier.cfg, which is just after the 'blobs' are deposited.

496          _BLOBIFIER_COUNT
497        {% endif %}
498      {% endfor %}
499    {% endif %}
500    {% if safe.tray or ignore_safe %}
501      G1 Z{tray_top + 1} F{travel_spd_z}
502      G4 P{pressure_release_time}
503    {% endif %}
504    {% if safe.brush or ignore_safe %}
505      BLOBIFIER_CLEAN
506    {% else %}
507      G1 X{brush_start} F{travel_spd_xy}
508    {% endif %}

There should be a G90 between line 499 and line 500, 

499    {% endif %}
          G90
500    {% if safe.tray or ignore_safe %}

otherwise is safe.brush is 'false' and ignore_safe is 'false' it is still in relative positioning mode when it hits the G1 at line 507.

It could also go after the 'else' in line 506, as the other paths have their own G90.

506    {% else %}
            G90
507      G1 X{brush_start} F{travel_spd_xy}
508    {% endif %}

(although for the BLOBIFIER_CLEAN path that depends on if the SAVE_GCODE_STATE and RESTORE_GCODE_STATE also include capturing/restoring positioning mode (IE: relative or absolute). I have not looked that up myself. If they do then the G90 should really go after line 499..

One workaround is to set the variable_force_safe_descend_height_until value to the max printing height of the printer, but this is not recommended as it disables all safety checks and you need to manage the "safe printing zones" yourself.

I am sorry, I recently rebuilt my PC and lost all of my setup, so I cannot submit a merge/update request via github for this.

moggieuk, 2025-04-29:
Thanks for the effort in debugging this.  I'm going to tag @Dendrowen who is Mr Blobifer and would probably be better suited than me to review/confirm.

On a quick look it does make sense to me but @Dendrowen is much closer to this logic than me..

Dendrowen, 2025-05-01:
Looks good to me. Thanks!

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

In v4 all the Blobifier moves go through _SAFE_G1, which always moves in absolute coordinates, so the brush/tray moves no longer get caught by the relative mode left over from purging.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk