moggieuk/Happy-Hare #706: bypass mode not working as expected.

cruzer619, 2025-05-07:
sending log with debug mode on.

bypass mode steps.

- home mmu
- enable bypass mode, ERCF v2 Block
- manually load filament
- send print job from orcaslicer, single filament profile.

issue after sending print job.

- mmu tries to home, can't as filament is already loaded.
- printer is paused.
- hit play anyway, mmu doesn't like it, and tries to eject filament.

[mmu (3).log](https://github.com/user-attachments/files/20090000/mmu.3.log)

cruzer619, 2025-05-07:
forgot to mention, to print my print job.  I had to disable mmu.

moggieuk, 2025-05-09:
So, if I understand correctly you manually loaded filament in the bypass gate for a single color print?  I'll look at the log, but if you did the following:
`MMU_SELECT_BYPASS` or `MMU_SELECT BYPASS=1`
but then didn't load the extruder with:
`MMU_LOAD` or `MMU_LOAD EXTRUDER=1`
HH will not know filament is loaded.  If you manually load the extruder you can correct state with:
`MMU_RECOVER LOADED=1`

Typically you would do the following:
```yml
MMU_SELECT BYPASS=1
manually feed filament to extruder entrance..
MMU_LOAD
```
The MMU_LOAD will default to just loading the extruder but will correctly set the state in HH.  The homing move should then not occur because the MMU is already homed.

Disabling the MMU is also always and option.

cruzer619, 2025-05-10:
ok that worked but I still got errors.  It paused at the errors, but hit play anyway and its printing.  Here is the log for this one.

[mmu.log](https://github.com/user-attachments/files/20143468/mmu.log)

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

Selecting the bypass and then running MMU_LOAD is the right sequence so Happy Hare knows the filament is loaded; that still works the same way in v4.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk