moggieuk/Happy-Hare #707: Help with using happy hare without filament sensors.

CoreXYForge, 2025-05-10:
I read somewhere that it is possible (but not recommended) to use happy hare without filament sensors. I would like to do this, but I cannot find out how. Can someone help me?

moggieuk, 2025-05-11:
There needs to be some sort of "gate" or "post_gear" sensor (or encoder) as a minimum.  This is necessary to be able to load and home the filament AT the MMU -- you cannot rely on movement alone -- it isn't accurate enough.

If the extruder has a toolhead sensor then there is no need to home at the extruder end of the bowden. The lack of a sensor (encoder, extruder entry sensor, or sync-feedback compression) will only limit the automatic calibration of bowden length.

If you don't have any type of sensor near the extruder you can set the `extruder_homing_sensor: none` however this will likely be unreliable because movement over a large distance is not accurate -- even a well calibrated BMG gear will have quite a deviation over a large travel distance.  You can try but everything will have to be super well calibrated.

CoreXYForge, 2025-05-11:
Its fine that I can't. Last night I overcame my lazinesss and installed my BLtouch, so now I have a Z endstop I can use.

closed 2025-05-11 by CoreXYForge