moggieuk/Happy-Hare #709: Internal error during connect: invalid syntax (mmu_test.py, line 109)

Guamelon, 2025-05-14:
QiDI Q1 Pro，Re-installing HH did not help. 
I have seen similar errors before, which were caused by incompatibility between HH and Python 2. But the manufacturer told me that the Klipper they are using is a customized version and cannot upgrade to the Python version on their own. Is there a compatible patch available.

Guamelon, 2025-05-15:
Q1 Pro has a  Python 3.7.3 environment

JesNoFear, 2025-05-21:
I just updated mine to the current and got the same error.
i had to roll back to may 3rd using
`git reset --hard 22b1562376fbc3ca14b83421a041e38c682538bd`

JesNoFear, 2025-05-22:
Rolling back the version only worked for a little bit, and was getting `'module' object has no attribute 'PrinterRail'` when printer went into emergency, and would not recover.

but now i am back to `Internal error during connect: invalid syntax (mmu_test.py, line 109)` and cant load up my printer.

[logs.zip](https://github.com/user-attachments/files/20378483/logs.zip)

farmercyst, 2025-05-23:
> Rolling back the version only worked for a little bit, and was getting `'module' object has no attribute 'PrinterRail'` when printer went into emergency, and would not recover.
>
> but now i am back to [...]

I had the same error.  I commented out lines 31 and 651 from Happy-Hare/extras/mmu/mmu.py and it seems to be working for me now. 

`#from .mmu_test           import MmuTest`
and 
` #_ = MmuTest(self)`

moggieuk, 2025-05-28:
Thanks.  I strive to continue to support Python 2.  Sometimes I have PR's that have not been validated on Python 2 as is this case.

I'll address but really, Python 2 was deprecated on Jan 1st 2020. There are no longer security patches so really ANY vendor still using it should be shot ;-)

burkfers, 2025-05-28:
@CooperGerman authored the changes to `mmu_test.py`, perhaps they have an immediate idea? I'm entirely unfamiliar with python2 and the differences to pyrhon3.

JesNoFear, 2025-05-31:
Just a side note, I am running python 3. And was having the issues.

closed 2025-06-02 by moggieuk

JesNoFear, 2025-06-15:
it seems to be working for me. I also found while working on a few other things that my default Python was .2x and not 3. I removed Python 2 and rebuilt the virtual env for Klipper.