moggieuk/Happy-Hare #715: Espooler-discard all movement for eject

53Aries, 2025-05-21:
Boxturtle. Currently when eject is performed the espooler is engaged to rewind the spool. This can cause the filament tip to be pulled out completely and become unconstrained. I propose remove all espooler for eject function. This would mean that there will be nothing to pull filament out of the pre gear switch though, so the switch would stay triggered until you manually remove the filament. I'm not sure if that could be an issue in code. Eject should be a move of E-{hub parking distance+gear teeth to post gear switch+some factor of extra to be sure filament is out of the gear}. Gear teeth to post gear switch is roughly 28mm for boxturtle extruder. To simplify and be more universal the move could simply be hub parking distance+user entered variable based on measurement or testing, or even ONLY a user entered variable for eject length as it will not hurt anything to have the eject length be higher than physically needed.

53Aries, 2025-05-21:
Disregard. After looking at HH code I discovered I simply had too much distance in the `gate_final_eject_distance`. HH currently ejects by homing to the post gear sensor then ejects the value of `gate_final_eject_distance` extra. As long as that value is a sane number the espooler will not pull filament out of the feeding bowden. So really no change is needed. Closing ticket.

moggieuk, 2026-10-23:
Thanks for raising this. Most of it was already solved by the home-to-sensor logic added in August 2025, after this issue was opened. When a gate has an entry (pre-gear) sensor, as Box Turtle does, the final eject now homes to that sensor rather than moving blindly. The move stops, and the espooler with it, as soon as the filament clears the switch, so the tip is no longer pulled out.

#1451 improves the remaining case. When a gate has no entry sensor, the final eject is still a blind move of `gate_final_eject_distance`. The espooler no longer rewinds during that move, so it can't keep pulling the filament off the gate once the gear has let go. The earlier part of the eject, backing off the exit sensor while the gear still grips the filament, keeps its rewind.

This will be available in the v4.2.0 release.

closed 2026-10-23 by moggieuk

moggieuk, 2026-10-23:
#1451 has been merged into `development` and will be in the v4.2.0 release. It stops the espooler rewinding on the final eject move when the gate has no entry sensor. If an espooler operation is already running, for example one started with `MMU_ESPOOLER`, the eject also turns it off.