moggieuk/Happy-Hare #723: FR: Have MMU run MMU_Home on gate change

JesNoFear, 2025-05-29:
I would like to have the option added in to run MMU_Home for each gate change, something that can be enabled / disabled. it would catch so many of the miss-loads that i am seeing due to filament not being completely ejected, and error handling can be coded for such instead of "assuming" it moved to the right gate, then reload the partially loaded filament and thinking everything is OK.

JesNoFear, 2025-05-31:
I have it enabled, but it doesn't seem to work well for me, it homed fine but wasn't detecting failed movements.

igiannakas, 2025-05-31:
Adjust the threshold - try holding the selector by hand and try selecting another gate. It should recognise its stalled and try again.

moggieuk, 2025-06-02:
I really don't think this is a good option -- really the fix should be to address why filament is getting trapped.  However you can force this behavior by turning on the "macro load/unload" feature and adding a "MMU_HOME" operation at the start of the load...

closed 2025-06-15 by JesNoFear