moggieuk/Happy-Hare #729: MMU issue detected. Print will be paused Reason: Runout detected on None EndlessSpool mode is off - manual intervention is required

Willpowerandstring, 2025-06-05:
Getting random pauses with the above message. MMU_RECOVER LOADED=1 and resume and we're back printing again, for a period ( could be 30 mins, or 2 mins , seems random ) 

[mmu.log](https://github.com/user-attachments/files/20611501/mmu.log)

Any one shed any light on this? 

I'm running a BTT board  on an ender3 frame, I've a sprite extruder and no sensors on tool head ( collision detection enabled ) 

It's a relatively new setup but was working for a couple of days without any issues & nothing changed config-wise apart from tweaking the motor current up to avoid stalling on loading.

igiannakas, 2025-06-10:
I see this in the log:
13:06:06   DEBUG: Filament not detected in encoder after buzzing gear motor

How is your encoder readings? Are they consistent? It is triggering a run out right now - basically its not moving while filament is feeding.

Willpowerandstring, 2025-06-11:
Partial filament blockage it seems - due to the design of the print and the huge number of retractions involved it was very difficult to tell - what is the runout detected on "none" bit about? That was confusing me somewhat. Why would it not report a gate# or is this how gate 0 is reported?

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

In v4 the runout message names the tool and gate (and the sensor that triggered it) instead of printing 'None'.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk