moggieuk/Happy-Hare #742: Cannot set spool id from mainsail

wormiedk, 2025-07-13:
I have setup the spoolman integration

mmu_parameters:
```
#
spoolman_support: push			# off = disabled, readonly = enabled, push = local gate map, pull = remote gate map
pending_spool_id_timeout: 20            # Seconds after which this pending spool_id (set with rfid) is voided
#
```

moonraker.conf
```
# https://moonraker.readthedocs.io/en/latest/configuration/#spoolman
[spoolman]
# URL to the spoolman instance. This parameter must be provided.
server: http://10.0.1.6:7912
# The interval, in seconds, between sync requests with the spoolman server.  The default is 5.
sync_rate: 5
```

MMU_SPOOLMAN SPOOLINFO=8 works:
```
// Spool is: Hvid (id: 8)
// - Material: n/a
// - Used: 978 g
// - Remaining: 21 g
// Spool id 8 is not assigned to this printer!
// Run: MMU_SPOOLMAN SPOOLID=8 GATE=.. to add
```

When I try to assign a spool to a gate the map is not saved:
```
10:39:05 
$ MMU_SPOOLMAN SPOOLID=8 GATE=0
10:39:05 
// Spool 8 assigned to printer K1SE-65CD @ gate 0 in Spoolman db
10:39:22 
$ MMU_GATE_MAP REFRESH=1
10:39:22 
// Gates / Filaments:
// 0: Buffer;  Id: n/a;    Unknown | 240°C | n/a | Unknown
// 1: Spool;   Id: n/a;    Unknown | 240°C | n/a | Unknown
// 2: Empty;   Id: n/a;    Unknown | 240°C | n/a | Unknown
// 3: Empty;   Id: n/a;    Unknown | 240°C | n/a | Unknown
```

​If I go to edit the spoolmap (tools - edit filaments), nothing happens when pushing save

[image]

wormiedk, 2025-07-28:
The problem was that nothing happened when pushing the "save" button on the fluidd interface at all. After switching to readonly everything works as expected... As soon as I set it to "Push" the save button stops working

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

From the later comments this looks like the Fluidd/Mainsail edit dialog refusing to save rather than Happy Hare; MMU_GATE_MAP GATE=0 SPOOLID=8 sets the spool directly in the meantime.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk