moggieuk/Happy-Hare #751: Uncontrolled Really High Accels and Inaccurate Pinky

AcrimoniousMirth, 2025-07-20:
Running Happy Hare v3.3.0-1-ga0ca2e4f.
Having issues with preloads and gate checks not fully unloading the selector (by a varying amount from several mm to the length of the selector). I was getting some help on the Discord but we couldn't figure out why the acceleration in the trace was in the thousands when my configs are all in the hundreds.

Test config readout:
```
FILAMENT MOVEMENT SPEEDS:
gear_from_spool_speed = 50.0
gear_from_spool_accel = 50.0
gear_unload_speed = 50.0
gear_unload_accel = 50.0
gear_short_move_speed = 50.0
gear_short_move_accel = 100.0
gear_short_move_threshold = 70.0
gear_homing_speed = 25.0
extruder_homing_speed = 18.0
extruder_load_speed = 16.0
extruder_unload_speed = 16.0
extruder_sync_load_speed = 18.0
extruder_sync_unload_speed = 18.0
extruder_accel = 400.0

TMC & MOTOR SYNC CONTROL:
sync_to_extruder = 0
sync_form_tip = 1
sync_purge = 1
sync_feedback_enabled = 0
sync_feedback_buffer_range = 8.0
sync_feedback_buffer_maxrange = 12.0
sync_multiplier_high = 1.05
sync_multiplier_low = 0.95
sync_gear_current = 40%
extruder_collision_homing_current = 30%
extruder_form_tip_current = 100%
extruder_purge_current = 100%

LOADING/UNLOADING:
gate_homing_endstop = encoder
gate_unload_buffer = 50.0
gate_homing_max = 70.0
gate_parking_distance = 23.0
gate_preload_homing_max = 70.0
gate_preload_parking_distance = 23.0
gate_autoload = 1
gate_final_eject_distance = 0.0
bypass_autoload = 1
bowden_apply_correction = 0
bowden_allowable_load_delta = 30
bowden_pre_unload_test = 1
extruder_force_homing = 0
extruder_homing_endstop = extruder
extruder_homing_max = 100.0
toolhead_extruder_to_nozzle = 90.6
toolhead_entry_to_extruder = 21.5
toolhead_residual_filament = 0.0
toolhead_ooze_reduction = 0.0
toolhead_unload_safety_margin = 10
toolhead_post_load_tighten = 0
toolhead_post_load_tension_adjust = 1
gcode_load_sequence = 0
gcode_unload_sequence = 0

TIP FORMING:
form_tip_macro = _MMU_FORM_TIP
force_form_tip_standalone = 1

PURGING:
purge_macro =
force_purge_standalone = 0

LOGGING:
[... 129 lines cut ...]
09:27:32     TRACE: Final parking. Stepper: 'gear' moved -13.8mm, encoder measured 23.1mm (delta -9.3mm). Pos: @58.2, (66.3mm)
09:27:32 Warning: Possible encoder malfunction (free-spinning) during final filament parking
09:27:32   DEBUG: Selecting tool T0 on gate 0...
09:27:32     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Selecting' OLD_ACTION='Checking'
09:27:32     TRACE: Setting servo to move (filament hold) position at angle: 155
09:27:32     TRACE: Positioning selector
09:27:32       STEPPER: SELECTOR MOVE: position=4.0, speed=200.0, accel=1200.0
09:27:32     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Checking' OLD_ACTION='Selecting'
09:27:32     TRACE: Setting gear motor rotation distance: 23.0022
09:27:32 Tool T0 enabled
09:27:32 Gate : | 0 | 1 | 2 | 3 | 4 | 5 | 6 | 7 |
         Tools: |T0 |T1 |T2 |T3 |T4 |T5 |T6 |T7 |
         Avail: | ? | ? | * |   |   |   |   | ? |
         Selct: |\ /|---------------------------- T0
09:27:32     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Idle' OLD_ACTION='Checking'
09:27:32     TRACE: Setting servo to up (filament released) position at angle: 170
09:27:34     TRACE: Processing idle_timeout 'ready' event
```

Of note:
```
09:27:27       STEPPER: GEAR MOVE: dist=0.8, speed=25.0, accel=1000.0, wait=False
09:27:27       STEPPER: GEAR MOVE: dist=-0.8, speed=25.0, accel=1000.0, wait=False
09:27:32       STEPPER: SELECTOR MOVE: position=4.0, speed=200.0, accel=1200.0
```

AcrimoniousMirth, 2025-07-20:
Binky definitely not as accurate as it should be, tests giving 2.8-2.9mm/pulse... about double what it should be
- BMG gear has an effective diameter of 23, or one rotation for 23mm pushed
- Encoder wheel has 12 tabs, so 24 transitions
- Base resolution is 0.958 mm/pulse (23mm over 24 pulses)

AcrimoniousMirth, 2025-07-28:
Thanks, I found the erroneous accels reading the configs closely again!

So the Binky thinks it has a resolution of 3-4mm (reading in the range of 100-120 pulses for the 400mm test calibration) but the encoder wheel is well seated, it turns freely and consistently when filament moves through it and doesn’t appear to slip let alone 3/4 of the time. The extruder gears also leave perfectly regular and spaced marks in the filament indicating no slipping or grinding. 
Interestingly I invested in a second Binky to ensure it wasn’t just a board malfunction and the second Binky gave the same results.

Everything’s set up correctly to the best of my knowledge, extruder gears to 23.00028 mm/rev and extruding exactly 100mm when 100mm is requested.

I’ll get you the encoder reading after work, gotta dash now (UK time as well).

AcrimoniousMirth, 2025-07-28:
> 2. What is the reading of the encoder before and after that move?  You can use `MMU_ENCODER` to see current value

Evening!
```
18:59:19   MMU_ENCODER
18:59:20   Encoder position: 321.1
- Runout detection: Disabled
- Clog/Runout mode: Automatic (Detection length: 10.0)
- Remaining headroom before trigger: 15.0 (min: 10.0)
- Flowrate: 0 %

18:59:34   MMU_TEST_MOVE move=100

19:03:26   Encoder position: 321.1
- Runout detection: Disabled
- Clog/Runout mode: Automatic (Detection length: 10.0)
- Remaining headroom before trigger: 15.0 (min: 10.0)
- Flowrate: 0 %
```

moggieuk, 2025-10-26:
Sorry, this is a very late follow up and you have probably solved the issue by now, but Binky has a resolution of close to 1mm.  If it is calibrating as 4mm then it is probably a mechanical problem -- maybe some slots of the slotted disc are blocked or the sensor is not reading correctly...

Experiment with binky build first.  You can manually pull filament through very carefully and observe led flashing indicating it detected movement

closed 2025-10-26 by moggieuk