moggieuk/Happy-Hare #754: Option to have sifferent load speeds for different filaments / gates

utzy79, 2025-07-26:
Absolutely loving this software, but the one thing I have an issue with on my tradrack is Silk PLA slipping while feeding.
It would be great if we could choose the load speed per gate, Std PLA and any non silk filaments can load alot more reliable than Silk does.

ningpj, 2025-07-26:
HH already supports this and allows you speed overiders per gate.  Use the ``MMU_GATE_MAP GATE=xxx SPEED=xx`` to manage this and reduce gate  speed as a % of ``gear_from_gate_speed``.  Settings are persisted in ``mmu_vars.cfg``.  For example I have 12 gates, gates 0-8 have buffers and run at full speed, gates 9-11 have filamentalists and run at 88% of 380mm/s max setting. If I have TPU loaded, I slow the gate down much further and usually have it set around 50%.

``mmu_state_gate_speed_override = [100, 100, 100, 100, 100, 100, 100, 100, 88, 88, 88, 88]``

ningpj, 2025-07-27:
Please close the issue if this resolves your question and encourage you to head over to the Happy Hare discord community for further questions, clarification, configuration options, etc. https://discord.gg/qfz8bhVh

closed 2025-07-27 by utzy79