moggieuk/Happy-Hare #776: Creality Ender 3 V3 KE

matthew73210, 2025-09-05:
Hey,

I managed to get this installed, but updates are always a pain. Basically, the KE has no LEDs, so in their wisdom, Creality doesn’t include the LED module. As a result, when you import HH configs via printer.cfg, Klipper throws a fit about the missing led Python files. This happens even if you select “No LED” in the installer.

If possible, could you patch in the missing LED files or add placeholder stubs to keep Klipper happy?

Cheers - love what you’ve built.

nicochi, 2025-09-05:
Hi, 

I am having the same issue with K1 (that has LED in enclosure). 

@matthew73210 advise how you fix this manually?

Error from fluidd is: 

```shell
Internal error during connect: cannot import name 'led' from 'extras' (/usr/share/klipper/klippy/extras/__init__.py)

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
``` 
and from Klyppy.log:

```shell
[ERROR] 2025-09-05 16:52:25,096 [root] [klippy:_connect:248] Unhandled exception during connect
Traceback (most recent call last):
  File "/usr/share/klipper/klippy/klippy.py", line 193, in _connect
    self._read_config()
  File "/usr/share/klipper/klippy/klippy.py", line 159, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/usr/share/klipper/klippy/klippy.py", line 117, in load_object
    mod = importlib.import_module('extras.' + module_name)
  File "/usr/share/klippy-env/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 749, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/share/klipper/klippy/extras/mmu_machine.py", line 31, in <module>
    from . import mmu_leds
  File "/usr/share/klipper/klippy/extras/mmu_leds.py", line 18, in <module>
    from . import led as klipper_led
ImportError: cannot import name 'led' from 'extras' (/usr/share/klipper/klippy/extras/__init__.py)
```

matthew73210, 2025-09-05:
The way i hack through the error is to pass the led.py file from the K1 in klippy/extras to the ender. I'm currently having other issues right now, a mismatch between klipper version and HH.

matthew73210, 2025-09-05:
FYI the latest HH V.3.4 isn't compatible with the klipper version. It fails with `{"code":"key60", "msg":"Internal error on command:MMU_PRINT_START", "values": ["MMU_PRINT_START"]} ` which didn't happen on V3.3

matthew73210, 2025-11-01:
Going to investigate this some more. I'll close the issue are recreate one using the new method announced on the 26/10/25.

closed 2025-11-01 by matthew73210