moggieuk/Happy-Hare #777: Installation on Creality K1 CFS (firmware 2.3.5.33)

nicochi, 2025-09-05:
Work in progress, not yet sucesfull...

Installation on Creality K1 firmware 2.3.5.33:

1) install bash from entware

```shell
/opt/bin/opkg install bash
```

2) run installer as:

```shell
/opt/bin/bash install.sh -i
```

do not eanble LED support

3) optional: install klipper-led-effect

see https://github.com/julianschill/klipper-led_effect/issues/254

4) Error: Error: cannot import name 'led' from 'extras'

Error from fluidd is: 

```shell
Internal error during connect: cannot import name 'led' from 'extras' (/usr/share/klipper/klippy/extras/__init__.py)

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
``` 
and from Klyppy.log:

```shell
[ERROR] 2025-09-05 16:52:25,096 [root] [klippy:_connect:248] Unhandled exception during connect
Traceback (most recent call last):
  File "/usr/share/klipper/klippy/klippy.py", line 193, in _connect
    self._read_config()
  File "/usr/share/klipper/klippy/klippy.py", line 159, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/usr/share/klipper/klippy/klippy.py", line 117, in load_object
    mod = importlib.import_module('extras.' + module_name)
  File "/usr/share/klippy-env/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 749, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/share/klipper/klippy/extras/mmu_machine.py", line 31, in <module>
    from . import mmu_leds
  File "/usr/share/klipper/klippy/extras/mmu_leds.py", line 18, in <module>
    from . import led as klipper_led
ImportError: cannot import name 'led' from 'extras' (/usr/share/klipper/klippy/extras/__init__.py)
``` 

I copy led.py from klipper version 0.11.0 from  https://github.com/Klipper3d/klipper/tree/v0.11.0/klippy/extras or use creality klipper repo. https://github.com/CrealityOfficial/K1_Series_Klipper/tree/main/klippy/extras (yes, old but is the version creality K1 uses) and disabled led support (answer n) in from installation script.

(if on firmware 2.3.5.33 you need to copy the display folder)

I choosed to copy on HappyHare and soft link. Your choice.

if LED support is enabled I get followng error (I believe becouse Klipper is outdated):

```shell
[ERROR] 2025-09-05 17:23:57,697 [root] [klippy:_connect:248] Unhandled exception during connect
Traceback (most recent call last):
  File "/usr/share/klipper/klippy/klippy.py", line 193, in _connect
    self._read_config()
  File "/usr/share/klipper/klippy/klippy.py", line 159, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/usr/share/klipper/klippy/klippy.py", line 126, in load_object
    self.objects[section] = init_func(config.getsection(section))
  File "/usr/share/klipper/klippy/extras/mmu_machine.py", line 1291, in load_config
    return MmuMachine(config)
  File "/usr/share/klipper/klippy/extras/mmu_machine.py", line 117, in __init__
    unit.leds = mmu_leds.MmuLeds(c, self, unit, unit.first_gate, unit.num_gates)
  File "/usr/share/klipper/klippy/extras/mmu_leds.py", line 85, in __init__
    self.virtual_chains[segment] = VirtualMmuLedChain(config, "unit%d" % self.mmu_unit.unit_index, segment, config_chains)
  File "/usr/share/klipper/klippy/extras/mmu_leds.py", line 30, in __init__
    self.led_helper = klipper_led.LEDHelper(led_config, self.update_leds, sum(len(leds) for chain_name, leds in config_chains))
  File "/usr/share/klipper/klippy/extras/led.py", line 23, in __init__
    white = config.getfloat('initial_WHITE', 0., minval=0., maxval=1.)
AttributeError: module 'extras.output_pin' has no attribute 'lookup_template_eval'
``` 

5) Error: No such file or directory: '/root/printer_data/config/mmu/mmu_vars.cfg'

```shell
[ERROR] 2025-09-05 17:54:56,467 [root] [klippy:_connect:248] Unhandled exception during connect
Traceback (most recent call last):
  File "/usr/share/klipper/klippy/klippy.py", line 193, in _connect
    self._read_config()
  File "/usr/share/klipper/klippy/klippy.py", line 159, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/usr/share/klipper/klippy/klippy.py", line 126, in load_object
    self.objects[section] = init_func(config.getsection(section))
  File "/usr/share/klipper/klippy/extras/save_variables.py", line 64, in load_config
    return SaveVariables(config)
  File "/usr/share/klipper/klippy/extras/save_variables.py", line 16, in __init__
    open(self.filename, "w").close()
FileNotFoundError: [Errno 2] No such file or directory: '/root/printer_data/config/mmu/mmu_vars.cfg'
```

create a softlink:

```shell
/root [#] ln -s /usr/data/printer_data/ ./
```

6) Error: gcode command T0 already registered

```shell
{"code": "key57", "msg": "gcode command T0 already registered\n\nOnce the underlying issue is corrected, use the \"RESTART\"\ncommand to reload the config and restart the host software.\nPrinter is halted\n", "values": ["T0"]}
```

```shell
[ERROR] 2025-09-05 18:02:05,297 [root] [klippy:_connect:231] Config error
Traceback (most recent call last):
  File "/usr/share/klipper/klippy/klippy.py", line 193, in _connect
    self._read_config()
  File "/usr/share/klipper/klippy/klippy.py", line 159, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/usr/share/klipper/klippy/klippy.py", line 126, in load_object
    self.objects[section] = init_func(config.getsection(section))
  File "/usr/share/klipper/klippy/extras/box.py", line 3, in load_config
    return(MultiColorMeterialBoxWrapper(config))
  File "klippy/extras/box_wrapper.py", line 3448, in extras.box_wrapper.MultiColorMeterialBoxWrapper.__init__
  File "/usr/share/klipper/klippy/gcode.py", line 140, in register_command
    raise self.printer.config_error(
configparser.Error: {"code":"key57", "msg":"gcode command T0 already registered", "values": ["T0"]}
```

nicochi, 2025-09-22:
Comment all the lines in mmu_hardware.cfg that has to do with LED, should be the one listed above. Add # in front. Sorry if that was not clear. 

Had not time to look into next steps yet, but now I got the hardware to continue. 

In parallell I am also trying a different approach of decoding Modbus 485 messages and interface directly with Creality firmware w/o need for HH. Some initial success at loading and unloading filament, but there is more work to do.

moggieuk, 2025-10-24:
I'm afraid I don't have access to a Creality machine to test.  I just pulled a PR that was offered to help with installation and setup (changes to install.sh).

I know Creality users are using Happy Hare so hopefully there is somebody that can help.

I will note that some of your errors seem to be incorrect `.cfg` files.  Here's a tip.. Run install.sh with these flags and a config will be created in /tmp.  You can compare that to what you have.
```
cd ~/Happy-Hare
./install.sh -k /tmp -c /tmp -i
```
(this will not touch your printer setup and is purely for testing / experimentation)

moggieuk, 2025-10-26:
Reading the issue again (and thanks for the details). I'm sensing two things:
1) an led incompatibility problem with very old klipper (Creality) -- possibly still exists?
2) a config issue where the "enabled" parameter from the next config section was being interpreted -- this is likely because an early section was not fully commented out (there was a bug in the installer that could have left a line uncommented because it depended on other user changes to that file).

Let me know where this stands now..

DrEvil23, 2025-11-16:
I had the same problem.  On the K1C, like on the Ender 3V3, you need klipper v12 source code.  Copy the led.py file from the source code and put the file in the folder "/usr/share/klipper/klippy/extras" of your Creality printer, using SSH.  Then edit the file using vim and comment out the line "from .display import display", and everything will start as it should, pretty sure you must not use the led modules though.

[image]

I realized it was Klipper V12 when I tried to make my 3MS system works, as it would not connect properly when compiled against Klipper V11.

closed 2025-12-31 by github-actions[bot]