moggieuk/Happy-Hare #786: FeatureRequest: Add Support for Anycubic ACE Pro

DerDoubleD, 2025-09-17:
This would be very nice!

Some informations:

Docu: https://github.com/ANYCUBIC-3D/Kobra3/blob/91d9771204b00242668a02586cf317a58a1aa26f/klipper-go/docs/N033%E6%96%99%E7%9B%92%E9%80%9A%E4%BF%A1%E5%8D%8F%E8%AE%AE.md

Some tests from me:
``
Get information about the box
curl -X POST -H "Content-Type: application/json" -d '{}' http://192.168.20.25:7125/printer/filament_hub/info

Get values about the box
curl -X POST -H "Content-Type: application/json" -d '{}' http://192.168.20.25:7125/printer/objects/query?filament_hub

Get information on consumables
curl -X POST -H "Content-Type: application/json" -d '{"id": 0, "index": 0}' http://192.168.20.25:7125/printer/filament_hub/filament_info
curl -X POST -H "Content-Type: application/json" -d '{"id": 0, "index": 1}' http://192.168.20.25:7125/printer/filament_hub/filament_info
curl -X POST -H "Content-Type: application/json" -d '{"id": 0, "index": 2}' http://192.168.20.25:7125/printer/filament_hub/filament_info
curl -X POST -H "Content-Type: application/json" -d '{"id": 0, "index": 3}' http://192.168.20.25:7125/printer/filament_hub/filament_info

Turn on drying-function
curl -X POST -H "Content-Type: application/json" -d '{"id": 0, "temp": 60, "fan_speed": 50.0, "duration": 2}' http://192.168.20.25:7125/printer/filament_hub/start_drying

Stop drying function
curl -X POST -H "Content-Type: application/json" -d '{"id": 0}' http://192.168.20.25:7125/printer/filament_hub/stop_drying

Get Feed box configuration
curl -X POST -H "Content-Type: application/json" -d '{}' http://192.168.20.25:7125/printer/filament_hub/get_config

Set Feed box configuration
curl -X POST -H "Content-Type: application/json" -d '{"auto_refill": 0, "flush_multiplier": 0.5, "flush_multiplier_editable": 0.9, "flush_volume_max": 801, "flush_volume_min": 105, "runout_detect": 1}' http://192.168.20.25:7125/printer/filament_hub/set_config
```

moggieuk, 2025-10-25:
I'd need someone with a Anycubic ACE Pro to build a PR for this. It's not something I can do without a printer.   If running klipper then HH should work.  The "install.sh" script may need to be tweaked ... I did just pull a PR for broadening support should it would be worth trying again with very latest patch of HHv3.4.1

bobsyouruncleswife, 2025-10-30:
I came here to make the same request, and it's already here.  :)  My ACE Pro just came in today and I have a Voron.  Is there a way I can help with this?  I'm not a coder, but I am a geek and work in IT.  Adding this to Happy Hare would be EPIC.  BunnyACE is the most current plugin, I think, for using the ACE Pro with Klipper and it is currently crashing my Moonraker.

bobsyouruncleswife, 2026-02-13:
Would it be possible for me to pick your brain on how you got it to work on your Voron?  I am struggling.