moggieuk/Happy-Hare #792: Option 'enabled' is not valid in section 'mmu_encoder mmu_encoder'

zeplintwo, 2025-09-25:
I am trying to track down how to correct this error on a MMB v2.0 CAN board swap. my old board died and need to be changed so i updated to the above board. but on the new install. I am stuck on this error 
`Option 'enabled' is not valid in section 'mmu_encoder mmu_encoder'`
 search does not produce anything actionable. I am running a ERCF v2 with the encoder cutter. 
i have attached the only place i can find that has the dual reference to the error but i can not find what it is talking about it being enabled. I clearly missed something in the documentation and made an error in configuration but i have no idea what it might be. my clipper logs has a bunch of errors for pre gate and gear sensors at the end of the log. but i have commented them out as i do not use them currently. I believe they are all optional as i have had this printer running without them but not with the newest version of HH. Please advise on actions.

[image]

[image]

[image]

zeplintwo, 2025-09-26:
I was looking at the logs abit more and found that the overruns into the next free section of LEDs and that is where the enable option is coming from not the actual mmu_encoder section. but when i enable the header for the leds to break it out of the mmu_encoder section. the leds fail. 

[image]
[image]

kotovasia5120, 2025-09-27:
I don't have LEDs, but the error is the same: ERCF v2 and CAN MMB v1.1

sakitume, 2025-10-06:
I just encountered this same issue with a new HappyHare install and I tried to perform a "Firmware restart" after customizing some of the parameters. Using google I came across this issue this issue and saw @kotovasia5120 post above and sure enough, in my `mmu_hardware.cfg` I see that many of the parameters in the `[mmu_leds]` section were *not* commented out even though the `[mmu_leds]` section was commented out.

My guess is that this is an issue with the install script when you specify you don't want any LEDs.